ARMmbed/stm32customtargets #8: PinNames.h of inherited board not used

hallard, 2021-07-12:
When trying to compile for board `LORA_E5_BREAKOUT` the associated file `PinNames.h` located into `TARGET_LORA_E5_BREAKOUT` is not used but it is the file located into `TARGET_LORA_E5` so the following definitions are missing at compile time and fire an error when trying to use LED1 for example.

```cpp
#define LED1     PB_5
#define LED2     PB_10
#define BUTTON1  PA_0
```

I don't know the best way to fix, we could remove inherit from LORA_E5 but in this case we need to duplicate files `LORA_E5_radio_driver.cpp` and `PeripheralPins.c` into target `LORA_E5_BREAKOUT`or I"m just missing a magic config elswhere so the compiler takes  `PinNames.h` from correct target folder?

Just let me know I'll do the PR

hallard, 2021-07-12:
I found another way, not best one but really easier on end user side and makes sense

remove the folder TARGET_LORA_E5_BREAKOUT (since the only file was `PinNames.h` and unused).

and just change `custom_targets.json` adding `macro_add` as follow

```json
    "LORA_E5_BREAKOUT": {
        "inherits": [
            "LORA_E5"
        ],
        "macros_add": [
            "LED1=PB_5", 
            "LED2=PB_10",
            "BUTTON1=PA_0"
        ]
    }
```

Works fine.

Just let me know

jeromecoutant, 2021-07-13:
Fine for me!

hallard, 2021-07-13:
ok done PR #9

PR #9 (2021-07-15): moved custom LED/Button on macro
Fix #8
closed 2021-07-13 by hallard