ARMmbed/stm32customtargets #19: LORA E5 - LPUART and other UART pins

jpmeijers, 2021-09-07:
https://github.com/ARMmbed/stm32customtargets/blob/e7db2bbc78af3db31695a252e8b9f243bc37d754/TARGET_STM32WL/TARGET_LORA_E5/PeripheralPins.c#L133-L159

I would like to use the LPUART on PC0+PC1. I've tried changing the definition to:
```
MBED_WEAK const PinMap PinMap_UART_TX[] = {
    {PA_2,       UART_2,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART2)}, // Connected to STDIO_UART_RX
    {PA_2_ALT0,  LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)}, // Connected to STDIO_UART_RX
//    {PA_9,       UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
//    {PB_6,       UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
//    {PB_11,      LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)}, // Connected to LED3
    {PC_1,       LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)},
//    {PA_2,       LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)}, // Connected to STDIO_UART_RX
    {PA_9,       UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
    {PB_6,       UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
    {NC, NC, 0}
};

MBED_WEAK const PinMap PinMap_UART_RX[] = {
    {PA_3,       UART_2,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART2)}, // Connected to STDIO_UART_RX
    {PA_3_ALT0,  LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)}, // Connected to STDIO_UART_RX
//    {PA_10,      UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
//    {PB_7,       UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
//    {PB_10,      LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)},
    {PC_0,       LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)},
//    {PA_3,       LPUART_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF8_LPUART1)}, // Connected to STDIO_UART_RX
    {PA_10,      UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
    {PB_7,       UART_1,   STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)},
    {NC, NC, 0}
};
```

This looks like it is working now, but it did not work the previous time. Maybe I had a mistake.

Can someone with more experience with the mbed custom target definitions please have a look and add the LPUART and  additional UART pins? @hallard

jeromecoutant, 2021-09-07:
I agree.
With NUCLEO_WL55JC, CONSOLE pins are using **LPUART_1**, so all others **LPUART_1** pins were commented.
With LORA_E5, CONSOLE pins are using **UART_1**, so all others **UART_1** pins should be commented, and **LP_UART1** available :-)

jpmeijers, 2021-09-07:
Is it necessary to comment out the pins used by the console? I've used printf and UnbufferedSerial on the same uart without issues.

For a module like the LORA E5, I guess it's up to the board designer which pins are used for the console, so all uart pins should be available.

hallard, 2021-09-07:
That's a good question, I was wondering yesterday which pins I could use for UART and if I could comment out some, 

From my point of view you could use uart only on pins defined here (and not commented out of course) but tell me if I'm wrong (it was like that on this [framwork](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/variants/B-L072Z-LRWAN1/variant.cpp) 

Console pins are defined on board template (makes sense), so by board designer but can be overwritten in `mbed_app.json`.

So if @jpmeijers uncomment works, let's do it. may be we could also use other pins, which would be great, but need to be tested.

PR #20 (2021-09-08): LORA_E5: correct pin table
Fix #19 

For information, used command to generate file templates:
```
python targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py -m "STM32WLE5JCIx.xml"
```
closed 2021-09-08 by jeromecoutant