ARMmbed/stm32customtargets #32: Assertion failed

prasannaad, 2022-06-23:
Hi,

I followed the[ instructions](https://github.com/hallard/LoRa-E5-Breakout) to use the LoRa E5 with Mbed Studio. But I am getting this error.

`++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: error_value == HAL_OK
Location: 0x800E97B
File: STM32WL_LoRaRadio.cpp+775
Error Value: 0x0
Current Thread: main Id: 0x20002658 Entry: 0x800F47D StackSize: 0x1000 StackMem: 0x20001080 SP: 0x20001F6C 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=LORA_E5_MINI
-- MbedOS Error Info --
`

I used the same code with RAK3172 and changed the custom target and it's working fine. 

[image]

jeromecoutant, 2022-06-28:
@hallard

hallard, 2022-06-28:
Just flashed my test prog on LoRa-E5 with fresh mbed-ide + mbed-os 6.16, no issue at all

```
Mbed LoRaWANStack initialized 
 CONFIRMED message retries : 3 
 Adaptive data  rate (ADR) - Enabled 
 Connection - In Progress ...
 Connection - Successful 
 Dummy Sensor Value = 3 
 23 bytes scheduled for transmission 
 Message Sent to Network Server 
 Dummy Sensor Value = 5 
 23 bytes scheduled for transmission 
```

jeromecoutant, 2022-07-22:
@prasannaad any update ?
Thx

prasannaad, 2022-07-22:
Hello @jeromecoutant and @hallard ,

It was a problem specific to that particular board. When I changed to a different board, it worked fine. 
Thank you!

closed 2022-07-22 by jeromecoutant