ARMmbed/stm32customtargets #35: question

tcpipchip, 2022-10-25:
[image]
Hi
can you help me about this error ?
Thanks
I followed your instructions 
https://github.com/hallard/RAK3172-Breakout#readme
thanks!!!

jeromecoutant, 2022-10-25:
Hi
Did you make this part:
> open the file custom_targets.json from folder stm32customtargets and copy whole contents

@hallard

tcpipchip, 2022-10-25:
Hi!!!
Good morning! Got progress!
I had tested on 4 computers in University, but some access rights should be the problem on computers.
got progress now in house!
[image]
Now i am having this error!
do you know that ?

jeromecoutant, 2022-10-25:
@0xc0170

hallard, 2022-10-25:
Could you pleas check in menu `help` then `check for tools updates` and paste version here?

Mire are
[image]

tcpipchip, 2022-10-25:
good morning Team
[image]

tcpipchip, 2022-10-25:
> Hi Did you make this part:
> 
> > open the file custom_targets.json from folder stm32customtargets and copy whole contents
> 
> @hallard

yes!

tcpipchip, 2022-10-25:
i will uninstall and install the MBED STUDIO...

tcpipchip, 2022-10-25:
[image]
Now compiles!
One more question, wich TARGET processor should i choose that use the STM32WL55?

hallard, 2022-10-25:
Nice,

Depends on the board you use, hard to answer without knowing your hardware

tcpipchip, 2022-10-25:
We are representative on South America of WISOL
Our Hardware use STM32WL55...
One of the list uses the STM32WL55 ?

hallard, 2022-10-25:
In this case why are you using custom targets? did you tried Nucleo STM32WL55?

tcpipchip, 2022-10-25:
now show me Murata, where can i add the Custom Murata ?

hallard, 2022-10-25:
`Murata Type ABZ` was the target selected when I've done the capture for you (it was the last I worked on). I clicked on the actual target (here `Murata Type ABZ`) that brings you target search list then just type target you need (in the capture `WL55`)

As far as I remember, no need custom target for `Murata Type ABZ`, it's built-in like `Nucleo-WL55JC`

tcpipchip, 2022-10-25:
yes, you are right! It´s WL55JC

tcpipchip, 2022-10-25:
[image]

tcpipchip, 2022-10-25:
i will see if will compiles the ABZ to STM32WL55

tcpipchip, 2022-10-25:
[image]

tcpipchip, 2022-10-25:
do you know where is the path to perifpheral info, uart, spi, etc of Nucleo-WL55JC?

tcpipchip, 2022-10-25:
getting progress here!
[image]

tcpipchip, 2022-10-25:
still not connect
working on sub-bands AU915

tcpipchip, 2022-10-25:
You help me a lot!
ok, only confirm wich class mbed use!
A,B or C ?

`  "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "mbed-trace.enable": false,
            "mbed-trace.max-level": "TRACE_LEVEL_DEBUG",
            "lora.over-the-air-activation": true,
            "lora.duty-cycle-on": true,
            "lora.phy": "AU915",
            "lora.device-eui": "{ 0x00, 0xX0, 0xE1, 0x15, 0xXX, 0xXX, 0xD8, 0x0X}",
            "lora.application-eui": "{ 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x99 }",
            "lora.application-key": "{ 0xFF, 0x9F, 0x13, 0xXX, 0x40, 0x18, 0x0A, 0xXX, 0x5D, 0x68, 0x46, 0xE0, 0xA0, 0x14, 0x69, 0x54 }"
        },`

and

 `"fsb-mask" = "{0xFF00, 0x0000, 0x0000, 0x0000, 0x0002}"`

tcpipchip, 2022-10-25:
here datasheet
https://github.com/SeongJiIoT/LSM1x0A_SDK_LoRaWAN104

tcpipchip, 2022-10-25:
here how is going the blog that i dont want to abort :)
https://lom204-cli-wisol.blogspot.com/2022/10/o-objetivo-deste-blog-e-mostrar-um-e.html

hallard, 2022-10-25:
It's up to you to choose your class depending on your device needs.
https://os.mbed.com/docs/mbed-os/v6.15/apis/lora-tech.html

jeromecoutant, 2022-10-26:
Issue is now followed with https://github.com/ARMmbed/mbed-os-example-lorawan/issues/243

closed 2022-10-26 by jeromecoutant

tcpipchip, 2022-10-26:
[image]
Working now on LOM204A :)